chipfoundry/openlane2 #674: GHLD Module not found

fred-corp, 2025-02-19:
### Description

When trying to use a VHDL source with the ``` --flow VHDLClassic``` argument, I get the following Yosys error :

```sh
ERROR: Can't load module `./ghdl':                                                                                               
                    dlopen(/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so, 0x0005): tried:               
                    '/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so' (no such file),                     
                    '/System/Volumes/Preboot/Cryptexes/OS/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plug             
                    ins/ghdl.so' (no such file),                                                                                                     
                    '/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so' (no such file)   ```

### Expected Behavior

GHDL plugin should be loaded with Yosys.

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
Failed to get Nix info: [Errno 2] No such file or directory: 'nix'
kernel: Darwin
kernel_version: 24.3.0
supported: True
distro: macOS
distro_version: 15.3.1
python_version: 3.11.9
python_path:
  - /nix/store/fgazxsglfa1iamspfj984vl3n76l7zm7-python3.11-openlane-2.3.8/bin
  - /nix/store/0hhzvkw889bsybhqxy12ky4jx6a95p2d-python3-3.11.9/lib/python311.zip
  - /nix/store/0hhzvkw889bsybhqxy12ky4jx6a95p2d-python3-3.11.9/lib/python3.11
  - /nix/store/0hhzvkw889bsybhqxy12ky4jx6a95p2d-python3-3.11.9/lib/python3.11/lib-dynload
  - /nix/store/0hhzvkw889bsybhqxy12ky4jx6a95p2d-python3-3.11.9/lib/python3.11/site-packages
  - /nix/store/jjrp8fzcacibh621cl9klw5yvbqc6476-python3-3.11.9-env/lib/python3.11/site-packages
  - /nix/store/fgazxsglfa1iamspfj984vl3n76l7zm7-python3.11-openlane-2.3.8/lib/python3.11/site-packages
  - /nix/store/qad8ymplf9aq4jxwm007wfsiwpmf95l5-python3.11-click-8.1.7/lib/python3.11/site-packages
  - /nix/store/r367wx2bi3gra6ikc1xa8iv061phdfii-python3.11-cloup-3.0.5/lib/python3.11/site-packages
  - /nix/store/63k20scvxwd200mjnx8hxrjklj0m0mq1-python3.11-pyyaml-6.0.1/lib/python3.11/site-packages
  - /nix/store/dp8qpjnz7kflqqpfxdycri5lb5r2gqaj-python3.11-yamlcore-0.0.2/lib/python3.11/site-packages
  - /nix/store/kjssj1yfgn2pm1kfrlym4qqd4a7rrkk3-python3.11-rich-13.7.1/lib/python3.11/site-packages
  - /nix/store/shh3xl0l5c2izimp610lkbvbgmbm4p4z-python3.11-markdown-it-py-3.0.0/lib/python3.11/site-packages
  - /nix/store/p3xzw8rg1lf6w6yhd3ipwv6r475x74n5-python3.11-mdurl-0.1.2/lib/python3.11/site-packages
  - /nix/store/dlvrsixwy8xz1lr2c4vq828659nv5a3l-python3.11-pygments-2.17.2/lib/python3.11/site-packages
  - /nix/store/kfibrjs2mwaq12h12lx5jmmfzbw6nivx-python3.11-requests-2.31.0/lib/python3.11/site-packages
  - /nix/store/yg6lw7apm174pzc8gl5xni75vyrbd70z-python3.11-brotlicffi-1.1.0.0/lib/python3.11/site-packages
  - /nix/store/vmq8580x9imwzjy61cjjmxfjps73lh2b-python3.11-cffi-1.16.0/lib/python3.11/site-packages
  - /nix/store/ir0ixl0yvpv4x7vf7ysl39z0410ksiq8-python3.11-pycparser-2.22/lib/python3.11/site-packages
  - /nix/store/ik2pw8i84jv5whx3kxyywbdh84rzn5yn-python3.11-certifi-2024.02.02/lib/python3.11/site-packages
  - /nix/store/cw9m8ibz3n35l6vizs9vnlxk4khw5vi9-python3.11-charset-normalizer-3.3.2/lib/python3.11/site-packages
  - /nix/store/j2l1ywgfp9zc5a4x4bmc8b1gvmcl0fab-python3.11-idna-3.7/lib/python3.11/site-packages
  - /nix/store/9wdx14v42vcvrnl5pliqp6qckfc8mbzn-python3.11-urllib3-2.2.1/lib/python3.11/site-packages
  - /nix/store/wrv8bajabpkmc6phq3czc7vrz173adhn-python3.11-pcpp-1.30/lib/python3.11/site-packages
  - /nix/store/sgnh9k4di8v48cnihw8vlvzl6n88gqjv-python3.11-volare/lib/python3.11/site-packages
  - /nix/store/vq5frh1k00r0zzyn9iklrr1k4p6lfscx-python3.11-httpx-0.27.0/lib/python3.11/site-packages
  - /nix/store/s3q5yy84nfx239ll4dqa3mfkbsv6azv2-python3.11-anyio-4.3.0/lib/python3.11/site-packages
  - /nix/store/3nl0zigzfwn3irr241qk884i6cclg74h-python3.11-sniffio-1.3.1/lib/python3.11/site-packages
  - /nix/store/rxjcs59dhbgwdxn0bdxxaaycki5j84sx-python3.11-httpcore-1.0.5/lib/python3.11/site-packages
  - /nix/store/w4gcp9l3b7vfhnkn39gp63p1vqzkyvw5-python3.11-h11-0.14.0/lib/python3.11/site-packages
  - /nix/store/zbwr6q3jgdzlgc248f0my23s9k9l6nd9-python3.11-zstandard-0.22.0/lib/python3.11/site-packages
  - /nix/store/57xrhrl9wlqida3p0rxhh7ph89n1037x-python3.11-truststore-0.8.0/lib/python3.11/site-packages
  - /nix/store/39g07na1shijvdnkr9wwnl8dk5nnrrvm-python3.11-aiohttp-3.9.5/lib/python3.11/site-packages
  - /nix/store/vb1ca4jz2n9c75kky9r0sc3xsr0731b0-python3.11-attrs-23.2.0/lib/python3.11/site-packages
  - /nix/store/l2qkhsrkrrigs4i78x7v7g65kjnliggs-python3.11-multidict-6.0.5/lib/python3.11/site-packages
  - /nix/store/nfr0i2c962c9k19pw8mrjxsc01wx8i8z-python3.11-async-timeout-4.0.3/lib/python3.11/site-packages
  - /nix/store/hzqqlma9lav56qnswl0mcvfl05m9xk80-python3.11-typing-extensions-4.11.0/lib/python3.11/site-packages
  - /nix/store/9qgc7a4916xwrya9m54hwmrahikh72zi-python3.11-yarl-1.9.4/lib/python3.11/site-packages
  - /nix/store/r19j7nlm7k8l1h9knh5p070ihkliprsp-python3.11-frozenlist-1.4.1/lib/python3.11/site-packages
  - /nix/store/mc4zw9pi2l3jqqrhv6z3cxvni3x4r5yw-python3.11-aiosignal-1.3.1/lib/python3.11/site-packages
  - /nix/store/j6zv5rz0pr1h4il6d7qxs5791sxj7ai2-python3.11-aiodns-3.2.0/lib/python3.11/site-packages
  - /nix/store/72d7rlmv10899dv7vj8p21l5na3pyc92-python3.11-pycares-4.4.0/lib/python3.11/site-packages
[... 105 lines cut ...]
ERROR: Can't load module `./ghdl': dlopen(/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so, 0x0005): tried: '/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so' (no such file),                                
'/System/Volumes/Preboot/Cryptexes/OS/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so' (no such file), '/nix/store/zxzmb2hdyjahy1lq6yglcx6sxy6zrddi-yosys-with-plugins/share/yosys/plugins/ghdl.so' (no such file)                                      
[23:58:09] ERROR    Subprocess had a non-zero exit.                                                                                                                                                                                                                                   step.py:1364
[23:58:09] ERROR    Last 10 line(s):                                                                                                                                                                                                                                                  step.py:1369
                     /----------------------------------------------------------------------------\                                                                                                                                                                                               
                     |  yosys -- Yosys Open SYnthesis Suite                                       |                                                                                                                                                                                               
                     |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |                                                                                                                                                                                               
                     |  Distributed under an ISC-like license, type "license" to see terms        |                                                                                                                                                                                               
                     \----------------------------------------------------------------------------/                                                                                                                                                                                               
                     Yosys 0.46 (git sha1 e97731b9dda91fa5fa53ed87df7c34163ba59a41, clang++ 16.0.6 -fPIC -O3)                                                                                                                                                                           [... cut ...]
                    '/Users/fredericdruppel/opt/vlsi-toolchain/tests/pm32/runs/RUN_2025-02-19_23-58-09/01-yosys-vhdlsynthesis/config.json', '--extra-in', '/Users/fredericdruppel/opt/vlsi-toolchain/tests/pm32/runs/RUN_2025-02-19_23-58-09/01-yosys-vhdlsynthesis/extra.json',                  
                    '--output', '/Users/fredericdruppel/opt/vlsi-toolchain/tests/pm32/runs/RUN_2025-02-19_23-58-09/01-yosys-vhdlsynthesis/led.nl.v']) failed                                                                                                                                      
[23:58:09] ERROR    OpenLane will now quit.
```

donn, 2025-02-20:
So I neglected to include the architecture in the environment report but if you're on an Apple Silicon Mac:

GHDL is only supported on x86-64 platforms because that's what the Ada compiler supports for Nix, unfortunately. Also, upstream does not release for aarch64 on either Mac OR Linux:

```
~% file ~/Downloads/ghdl-macos-11-llvm/bin/ghdl
/Users/donn/Downloads/ghdl-macos-11-llvm/bin/ghdl: Mach-O 64-bit executable x86_64
```

Fortunately, if you have Rosetta installed, there's a workaround. You may type `nix develop .#devShells.x86_64-darwin.default` instead of `nix-shell`, which will use the x86-64 version of Yosys with the GHDL plugin (at a performance penalty.)

---

Can you confirm if you're on Apple Silicon?

fred-corp, 2025-02-20:
Yes, Running on Apple Silicon. `yosys -m ghdl` works for FPGA synthesis (with ICE40 toolchains, and oss-cad-suite) but doesn't in the nix-shell for VLSI synthesis.

After running `nix develop .#devShells.x86_64-darwin.default` in my openlane2 installation folder, I can happily run `openlane --flow VHDLClassic config.json` and the workflow seems to run smoothly. I'm also able to check the layout with `openlane --last-run --flow openinklayout config.json`

donn, 2025-02-20:
If the oss-cad-suite builds for macOS/aarch64 I think I may be able to monkey-patch their shared object to work with Nix. I'll look into it.