chipfoundry/openlane2 #682: Will Efabless keep supporting openlane2?

0xharry, 2025-03-06:
and any other related opensourced repo?

donn, 2025-03-06:
We're not sure what Efabless's plans are. The OpenLane team has already forked all relevant repos to github.com/aucohl.

We're trying to talk to Efabless to get these repos transferred so we can keep issue numbers and such. We recommend joining https://fossi-chat.org for more updates.

Thank you.

0xharry, 2025-03-07:
> We're not sure what Efabless's plans are. The OpenLane team has already forked all relevant repos to github.com/aucohl.
> 
> We're trying to talk to Efabless to get these repos transferred so we can keep issue numbers and such. We recommend joining https://fossi-chat.org for more updates.
> 
> Thank you.

Thank you for the reply. It's great to see the open-source silicon projects continuing to get support. Appreciate all the effort!

donn, 2025-11-19:
So great news on two accounts:

- ChipFoundry has acquired Efabless IP, including this repo. I reminded them to move this repo over to their organization. We'll set up a soft redirect to LibreLane in this repo's readme- but this should funnel Google Searches to ChipFoundry and LibreLane.
- ChipFoundy has adopted LibreLane, which is live and has a stable release now: https://github.com/librelane/librelane

donn, 2025-11-19:
Also, I have write access here again. :)