chipfoundry/openlane2 #687: Incorrect frequency calculation in clock period example

vamshiKrishnaReddy20, 2025-05-22:
### Description

The OpenLane2 documentation includes an example of calculating frequency from the clock period:

f = 1 / CLOCK_PERIOD = 1 / 25 ns = 25 MHz

[image]

However, the final frequency value is incorrect. Since 25 ns corresponds to 40 MHz (not 25 MHz), this example is misleading and may confuse users.

### Proposal

Update the example to correctly show the frequency calculation:

f = 1 / CLOCK_PERIOD = 1 / 25 ns = 40 MHz

This correction will align the documentation with the correct physics and help users better understand clock frequency and period relationships.

WrathofBhuvan11, 2025-05-31:
Hey I have a question on the setup.

vamshiKrishnaReddy20, 2025-05-31:
What's that

vamshiKrishnaReddy20, 2025-05-31:
What's that